Digilent/vivado-boards #3: PYNQ-Z1 board file

ekiwi, 2016-10-04:
It would be great if you could add a board file for the new [PYNQ-Z1](http://store.digilentinc.com/pynq-z1-python-productivity-for-zynq/).

sbobrowicz, 2016-10-04:
That board is primarily being supported by the PYNQ program, so we weren't planning on it, but your request is noted. If we get some time we'll put this together. I'll keep this issue open as a reminder.

For now, just use the board files in this repo called arty-z7-20. Those are entirely compatible with the PYNQ-Z1.

ekiwi, 2016-10-04:
Great. Thank you.

xbug00, 2016-10-19:
Aha !  I'd missed that one (arty-z7-20).  Thank you very much indeed.

brad0taylor, 2016-10-25:
Is the arty-z7-20 the same board as PYNQ?  What are the differences?

sbobrowicz, 2017-02-02:
They are nearly hardware equivalents, just the arty z7-7020 is smaller, does not have a power switch, and doesn't have the PDM microphone.  The Arty Z7-7020 has our normal suite of examples/docs/tutorials, where the PYNQ-Z1 only directs people to the PYNQ project. That said, all Arty Z7-7020 materials will run on a PYNQ-Z1 due to their similarities.

Closing this issue because it is not a bug and the feedback has been captured.

closed 2017-02-02 by sbobrowicz

CUinNYC, 2017-03-26:
At the very end, did anybody find/create a configuration file for the PYNQ-Z1?

jonathan-reynolds, 2017-04-07:
I would very much appreciate this too. Thanks.

Nuhoisen, 2017-10-30:
Is it possible to use the PDM microphone if my PYNQ project uses with the ARTY z7-7020 board file?

ekiwi, 2017-11-01:
Since people seem to keep on finding this issue:
There seems to be an official [PYNQ-Z1 board file from Xilinx](https://github.com/cathalmccabe/pynq-z1_board_files) now. (source: [PYNQ v2.0 documentation](https://pynq.readthedocs.io/en/v2.0/overlay_design_methodology/board_settings.html))