Digilent/vivado-boards #8: Zybo Z7 preset files invalid syntax.

jbootsma, 2018-02-19:
It seems the most recent versions of the zybo z7 preset.xml have some invalid syntax. When running board automation in a new block design the ps7 doesn't have any peripherals mapped. Rolling back to an older version of the preset files seems to make things work again.

sbobrowicz, 2018-02-21:
Hmmm... this commit looks like it has questionable syntax: 56826bab9fdfdc95abed732155f33b6b2508aef4 .  @elodg , I think the syntax should actually match the line 2 lines up.  The presets file has different syntax as the exported XMLs from Vivado, which is an official pain in the ***. Can you verify?

elodg, 2018-02-21:
Confirmed syntax error. Fixed: 7ed4cb6efb39f7e5caec1758feb6ab1deaa9ad77

closed 2018-02-21 by elodg