Digilent/vivado-boards #20: Zybo Z7-20: Vivado 2018.3 reports critical warnings in DDR interface

krjdev, 2019-05-13:
Hello,

Don't know if it's really an issue in the configuration. I'm a newbie with Vivado and ZYNQ devices.
Vivado 2018.3 reports four criticial warnings when I validate the design: 

> [PSU-1]  Parameter : PCW_UIPARAM_DDR_DQS_TO_CLK_DELAY_0 has negative value -0.050 . PS DDR interfaces might fail when entering negative DQS skew values. 
>  [PSU-2]  Parameter : PCW_UIPARAM_DDR_DQS_TO_CLK_DELAY_1 has negative value -0.044 . PS DDR interfaces might fail when entering negative DQS skew values. 
>  [PSU-3]  Parameter : PCW_UIPARAM_DDR_DQS_TO_CLK_DELAY_2 has negative value -0.035 . PS DDR interfaces might fail when entering negative DQS skew values. 
>  [PSU-4]  Parameter : PCW_UIPARAM_DDR_DQS_TO_CLK_DELAY_3 has negative value -0.100 . PS DDR interfaces might fail when entering negative DQS skew values. 

Are these values correct and can I ignore these warnings?
. 
Thanks in advance!

krjdev, 2019-05-14:
In addition:
[image]

elodg, 2019-05-14:
https://reference.digilentinc.com/reference/programmable-logic/zybo-z7/reference-manual#hardware_errata

closed 2019-05-14 by elodg

krjdev, 2019-05-14:
Okay, didn't noticed that. 

Thank you!