Digilent/vivado-boards #36: Zybo board files error on Ubuntu

1c3t3a, 2022-02-09:
Hi there!

I am running Vivado 2021.2 (64-bit) SW Build 3367213 on an Ubuntu 20.04 x86_64 PC and run into the following error message (extracted from vivado.log) when parsing the board_files provided in this Repo:

```
WARNING: [Board 49-26] cannot add Board Part digilentinc.com:zybo-z7-10:part0:1.0 available at /tools/Xilinx/Vivado/2021.2/data/boards/board_files/zybo-z7-10/A.0/board.xml as part xc7z010clg400-1 specified in board_part file is either invalid or not available
```

I try to setup a Zybo-Z7-10 board. Sadly the error is the same for all Zybo boards, only the Artan and Spartan boards are available within Vivado.

artvvb, 2022-02-09:
Hi Bastian,

Please confirm that you have support for Zynq-7000 devices installed. The attached screenshot is from the Help -> Add Design Tools and Devices menu, which reruns the installer.

[image]

Thanks,
Arthur

1c3t3a, 2022-02-09:
Hi Arthur, that actually did the job, the installer downloaded the board support successfully thanks a lot! :)

closed 2022-02-09 by artvvb